google/skywater-pdk #397: Segmentation fault while opening Magic VLSI with sky130 technology

ddm2000, 2022-10-02:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version:
- Platform:

RTimothyEdwards, 2022-10-02:
For one, you can provide enough information that somebody like me can diagnose the issue.  "segmentation fault while opening magic" is not actionable information.

pgielda, 2022-10-02:
Looking quickly at the logs seems that "make install" step failed. It says that your filesystem is read-only.

RTimothyEdwards, 2022-10-02:
I think you have more than one issue here.  The most problematic issue is that your version of magic is very old.  That is probably what is causing the error with `make`, although it would be necessary to look through the log file to figure out where the actual error is being reported.  Secondly, the `OSError ...  Read-only file system` message is a bit concerning, and probably has something to do with the VM;  do you really have sudo privileges within the VM?  Lastly, you should not use the source template file `sky130.tech` on the command line to magic.  This is a template file and not an actual tech file.

proppy, 2022-10-21:
@ddm2000 if you want to install the latest version of magic without building is from source, you could give a try to the conda package published here: https://anaconda.org/conda-forge/magic

closed 2022-10-21 by proppy