google/skywater-pdk #417: Running ngspice for sky130 process and open_pdks

emfhasan, 2023-04-06:
I am trying to use the ngspice simulator with xschem using skywater130 process and open_pdk. I got the following error when I was running one test example:
Error: Could not find include file ~/home/eda_work/eman/hassan/skywater-pdk/libraries/sky130_fd_pr_ngspice/latest/corners/tt.spice

I followed the instruction in the following video for [Stefan Schippers](https://www.youtube.com/@stefsport2002) to install the open_pdk, xschem for sky130 process and ngspice: https://www.youtube.com/watch?v=jXmmxO8WG8s

Any advice in this regard?

MOS5500, 2024-07-24:
Have you resolved? I have the same problem...

Kreijstal, 2024-12-02:
Did you guys solved it?

RTimothyEdwards, 2024-12-02:
You appear to be using the original skywater-pdk sources, not the open-pdks installation.
You may want to simplify things by using the pre-built PDK from volare:  https://github.com/efabless/volare
Otherwise you should follow the instructions in http://opencircuitdesign.com/open_pdks.
Under no circumstances should you use the files from https://github.com/google/skywater-pdk which were never patched with fixes and have been archived.