google/skywater-pdk #425: VCS error due to defining default net type as none for primitives

M0stafaRady, 2023-06-12:
## Expected Behavior
VCS compile the library Verilog files  

## Actual Behavior
VCS through erros like this for primitives 
```
Identifier 'SET' has not been declared yet. If this error is not expected, 
  please check if you have set default_nettype to none. 
``` 
line [46](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/models/udp_dff_nsr/sky130_fd_sc_hd__udp_dff_nsr.v#LL46C9-L46C9)

This error shows only with primitives. Compilation passed after commenting out the `default_nettype none` line before primitives. 

## Steps to Reproduce the Problem

1. compile the primitives with VCS

## Specifications

- Version:  T-2022.06_Full64
- Platform:  Chronologic VCS (TM)

amm-efabless, 2024-06-04:
Is it safe to comment out `` `default_nettype none `` and why does this help? I'm guessing it's a VCS quirk. I'm also not sure why `SET` is flagged but other ports/identifiers are not. `SET` seems to be declared as one would expect, but maybe this is a VCS bug or some other issue to do with `SET` often being a keyword in various implementations...?

M0stafaRady, 2024-06-04:
It works because the default net type in verilog is wire. So if the default net type doesn't change any net with no definition would considered as a wire. 
The 2 code below would be equivalent:
```
    input  SET  ;
```
```
    input  wire SET  ;
```
but they are not in the case of default net type none

amm-efabless, 2024-06-04:
Ah right, thanks @M0stafaRady! So I assume the reason it isn't flagged on [the line above it](https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/models/udp_dff_nsr/sky130_fd_sc_hd__udp_dff_nsr.v#L45) (`output Q`) is because later `Q` *is* declared as a `reg`. Cool, thanks for the feedback.