google/skywater-pdk #432: check_density.py script does not load topcell

mhommelga, 2023-10-04:
## Expected Behavior
$PDK_ROOT/sky130A/libs.tech/magic/check_density.py line 132:
"print('load ' + gdsroot, file=ofile)"
In order to output command to load topcell into the .tcl script

## Actual Behavior
"print('load ' + gdsroot)"
ie, does not output command to tcl script.

## Steps to Reproduce the Problem

1. All cells are reported as less than 700um x 700um and script exits regardless of actual cell size
1. Check check_density.tcl (use -keep option)
- Line after "gds read xxx.gds" should be "load xxx", but this is missing

## Specifications

- Version: ??
- Platform: Ubuntu

mithro, 2023-10-04:
You will want to report this to `https://github.com/RTimothyEdwards/open_pdks`

RTimothyEdwards, 2023-10-04:
@mithro :  Thanks for the heads-up.
@mhommelga :  Thanks for the bug report.  I see the error and will fix it in open_pdks.

mhommelga, 2023-10-05:
Thanks for the quick response - much appreciated. Sorry for posting in the wrong place