google/skywater-pdk #434: Where can I find the raw measurement data used to create the spice models for sky_130_fd_pr?

JamesTimothyMeech, 2024-04-04:
I am running Monte Carlo simulations in ngspice using  sky_130_fd_pr devices. When running a Monte Carlo simulation I see that some parameters of the spice models are sampled from a Gaussian distribution. Is the measurement data that was used to fit these Gaussian distributions to the parameters available online? If so where can I access these raw measurements?

QuantamHD, 2024-04-04:
The original data has been lost to time. These are simply what the foundry provided. We did have some of the devices recharacterized, and the data for that is here https://github.com/google/skywater-pdk-sky130-raw-data

No compact model was ever fitted to the new data.

JamesTimothyMeech, 2024-04-04:
Thanks! I will take a look at this!

closed 2024-04-04 by JamesTimothyMeech

JamesTimothyMeech, 2024-04-04:
Am I correct to conclude that this repo: https://github.com/google/skywater-pdk-sky130-raw-data includes one sweep for each device? If we wanted to fit Gaussians to measurements from many different instances of devices of the same type we would have to wait for the measurements to come back from something like this: https://github.com/google/skywater-pdk-sky130-raw-data/tree/main/docs/sky130-testtile-open?