google/skywater-pdk #436: Error to generate the timing lib for sky130_fd_io.

chchzh1978, 2024-07-12:
## Expected Behavior
Cell libs is generated.

## Actual Behavior
yiyang@yiyang-System-Product-Name:~/synopsysinstall/skywater-pdk$ make timing
echo sky130_fd_io
sky130_fd_io
Traceback (most recent call last):
  File "/home/yiyang/synopsysinstall/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/yiyang/synopsysinstall/skywater-pdk/env/conda/envs/skywater-pdk-scripts/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/yiyang/synopsysinstall/skywater-pdk/scripts/python-skywater-pdk/skywater_pdk/liberty.py", line 1178, in <module>
    sys.exit(main())
  File "/home/yiyang/synopsysinstall/skywater-pdk/scripts/python-skywater-pdk/skywater_pdk/liberty.py", line 1122, in main
    lib, corners, all_cells = collect(libdir)
  File "/home/yiyang/synopsysinstall/skywater-pdk/scripts/python-skywater-pdk/skywater_pdk/liberty.py", line 235, in collect
    assert corners, library_dir
AssertionError: libraries/sky130_fd_io/latest

## Steps to Reproduce the Problem
1. Change the Makefile, replace the definition for "SC_LIBS"
         #SC_LIBS = $(sort $(notdir $(wildcard libraries/sky130_*_sc_*)))
         **SC_LIBS = $(sort $(notdir $(wildcard libraries/sky130_fd_*)))**
2. make timing

## Specifications

- Version: latest
- Platform:

RTimothyEdwards, 2024-07-12:
@chchzh1978 :  Use https://github.com/efabless/skywater-pdk-libs-sky130_fd_io, where the liberty files have already been generated and you don't have to do "make timing", or use https://github.com/efabless/volare, or use https://github.com/RTimothyEdwards/open_pdks.  The Efabless updates apply countless patches to the I/O library that are not present in the Google version of the repository.

chchzh1978, 2024-07-17:
thx for your reply!
more question, sky130_fd_pr and sky130_fd_pr_reram is the same issue, where can i find them.

closed 2024-07-28 by chchzh1978