google/skywater-pdk #437: MOS Intrinsic Parameters

Camisad0, 2024-08-30:
Hi,

I want to ask how could I see the MOS intrinsic parameters for Sky130 PDK? For example the Vth and Kp of 01v8 MOS. I need these parameters for circuit design activities. I am currently not seeing any available documentations for this.

Thanks.

Alex-detailer, 2024-09-15:
Hi Camisad0 , are you using ngspice? 
If so , device parameters are accessible by the following object : @transistor_name [ param ]. 
For example if i want to print the gm of pmos transistor , i use .print @m.xm3.msky130_fd_pr__pfet_01v8 [gm]. 
If you are not sure about transistor name use display all command.

Hope it helps
Alex

probstcj, 2025-03-01:
Hi Casmisad0,

There is a way to do it if you know the proper equations. Using the `display all` command, and then `display <FET_NAME_UNDER_TEST>`, we unfortunately don't see the KP or anything related to it. However, if you use the following circuit and instructions, you can calculate it for yourself:

[image]

So, essentially, if we use this circuit, and load VG (V2 in this case) with a barely sub-threshold load (must be less than VTH which is given using `print <FET_NAME_UNDER_TEST>[vth]`), and measure the current, this gives us ID0. This is an important current for most equations.

Now, if we change VG to 0, which is far below the threshold voltage, and measure the current again, we will call this i_ds1. Using the Effective Voltage equation and the equation for sub-threshold current for FETs:

 $$V_{eff}=2nU_T\big[\frac{1}{\alpha}\big]$$
 $$i_{ds}=I_{D0}\cdot S\cdot e^{\frac{V_{eff}}{nU_T}}$$

We can rearrange them to solve for the sub-threshold slope factor (n):

$$n=\frac{V_{eff}}{U_T\cdot \ln\Big(\frac{i_{ds1}}{I_{D0}\cdot S}\Big)}$$

This is important in another equation, which is the triode equation. We want the FET to be in triode mode so we don't have to worry about the early voltage, and it is more predicatable:

$$i_ds=\frac{S\cdot K_{pn}}{n}\Bigg[(V_{eff}V_{ds})-\frac{{V_{ds}}^2}{2}\Bigg]$$

If we now measure a current in triode mode, we can call this i_ds2, we can solve this equation for K_PN,

$$K_{pn}=\frac{i_{ds2}\cdot n}{S\cdot (V_{eff}\cdot V_{ds} - {V_{ds}}^2)} $$

For the 1.8V NFET, I got 245uA/V^2 and for the 1.8V PFET I got 62.8uA/V^2

Let me know if you have any other questions!