google/skywater-pdk #444: Problems when dealing with resistors on Xschem and on Magic

lbraga740n, 2025-05-04:
Whenever I instantiate a resistor on Xschem the sheet resistance is wrong, for a 430.7k resistor I should have an aspect ratio of 215.35, instead I get 1114, as shown in the figure below.

[image]

The same behavior is found when importing this netlist to Magic, so after doing parasitic extraction and running my final simulations I am getting weird results and I cannot say for certain if this issues are due to the buggy resistor behavior. My layout has the correct sizing (using the 2000 ohm/sq sheet resistance as specified by the PDK documentation), and the resistor devices are created within Magic, I am not using the ones created when the netlist is used to generate the .mag file. On the figure below, the resistor imported form Xschem is on the left and the one created on Magic is on the right (with the correct sheet resistance)

[image]

Thank you for your help!

RTimothyEdwards, 2025-05-04:
You appear to be confusing `res_high_po` (319.8 ohms/square sheet resistance) with `res_xhigh_po` (2000 ohms/square sheet resistance).

lbraga740n, 2025-05-04:
Thank you for your help!